SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE- #10: may i use it with stm32? or what i should change if i use it with stm32?

123LY321, 2019-09-29:


SensorsIot, 2019-09-29:
Maybe you can use it with an STM32. I do not know because I do not use these chips.

closed 2019-09-29 by SensorsIot

123LY321, 2019-09-30:
Emm....Actually, I would like to know what the INT pin level is with and without radioactive seeds. Thank you

123LY321, 2019-09-30:
When using stm32, I calculated CPM by using pull up input and drop edge to trigger interrupt. Unfortunately, I could not interrupt at all. I tried to simulate GM with the key in the software without any problem,So I would like to ask you what the level of INT of this circuit is when there is radioactivity,thanks.

SensorsIot, 2019-09-30:
Maybe you watch around https://youtu.be/K28Az3-gV7E?t=442 . There you see the output voltage. I would try it without a pullup because I said, that the output has a high impedance.