SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE- #12: Error compiling

Tom-Neverwinter, 2019-12-31:
This code is very picky about what libraries and boards it uses. It refuses to compile without the proper libraries or board. Its disheartening that I am back to testing every library and board combination again. 

Which versions of the libraries?

Which boards are actually compiling? [this code DOES NOT compile on all esp32 boards as I discovered in the previous version.]

https://pastebin.com/iPvpbwfG [Wemos lolin 32] [as recommended]
https://pastebin.com/jVVkbUX1 [lolin32]
LOLIN D32 PRO
DOIT ESP32 DEVKIT V1

SensorsIot, 2019-12-31:
You need to use the 5.x version of the json library which was current when the video was made

SensorsIot, 2020-09-17:
I changed the code considerably. Maybe you try now?

closed 2020-09-17 by SensorsIot