SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE- #13: Program not working

leopardx7, 2020-07-08:
I bought the same boards used in the video on Youtube, I loaded the arduino code into the esp32 board, but the diplay remains stuck on "Measuring". It connects correctly to the ThingSpeak channel, but always linearly increasing measurements. The number grows continuously (view the attached image). I tried to insert some currentMillis and previousMillis println. Both variables are always 0 in the serial monitor.
So I tried adding currentMillis = millis (); at the beginning of the loop (), at this point the display has finally shown "radioactivity" with the usual increasing numbers. Even if the card is disconnected from the geiger counter, the data are the same and always increasing. How can I solve it?

[image]

mateusz-szafraniec, 2020-09-16:
Hi,
have look at line 189: `cpm = counts / minutes;`
minutes aer constant, 1
here is a problem.

SensorsIot, 2020-09-17:
I changed the program. Maybe you try it now. Input pin is on 17. Maybe you have to adapt it.

closed 2020-09-17 by SensorsIot