SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE- #14: HW bug?

mateusz-szafraniec, 2020-09-23:
Hi, did you tested this device?
For example manual counting of beeps comparing with counts from program?
IMHO device counts one beep as 3 or more.

Why?
Because this:
https://imgur.com/Dda5uIb
Voltage on "INT" output

Voltage on Q3 base:
https://imgur.com/D6RWXhu

I changed R19 to 1kOhm and interrupt to RISING, without this program counts none.

The true output should be on pin3 of NE555 - same pin that driving LED and Q4.

What do you think?

SensorsIot, 2020-09-23:
If this happens you can try to add a small delay here:

void IRAM_ATTR ISR_impulse() { // Captures count of events from Geiger counter board
  counts++;
  delay(50);
}

Increase this value till double counting stops

mateusz-szafraniec, 2020-09-24:
Loudspeaker and led works perfectly.
Time scale is 100uS/div
Voltage scale is 2V/div (INT) 0,2V/div (base)
I think that those oscillations came from HV DC/DC inverter - please look at spikes on Q3 base, spike is every 50uS - 20kHZ
Oscillations on INT output are at the same frequency.

You shouldn't use delay in an ISR!
`Interrupts must not call delay() or yield(), or call any routines which internally use delay() or yield() either.`
https://arduino-esp8266.readthedocs.io/en/latest/reference.html?highlight=interrupt#interrupts

SensorsIot, 2020-09-24:
This is a hardware plroblem. The software expects a denounced signal

closed 2020-09-24 by SensorsIot